Palakis/esphome-native-hdmi-cec #3: Unable to get the TV to turn on after power-cycling it

iDVB, 2024-02-23:
I have this simple code working and using HomeAssistant to send in...
```
cec_destination = 0
cec_data = [0x04]
```
...and its working to get the tv to turn on, but only when the tv is in standby mode. I'd really like to be able to turn the tv on from complete power-cycle as well. I only wired up 2 of the 3 (3rd was said to be optional) wires suggested in your readme. Perhaps 3 might work if the TV knows a device is there?

```yaml
services:
    - service: hdmi_cec_send
      variables:
        cec_destination: int
        cec_data: int[]
      then:
        - hdmi_cec.send:
            destination: !lambda 'return static_cast<unsigned char>(cec_destination);'
            data: !lambda 'std::vector<unsigned char> charVector; for (int i : cec_data) { charVector.push_back(static_cast<unsigned char>(i)); } return charVector;'
```

I can't believe I've been all over the place and can't seem to find a "programmatic solution to turn the tv on and off that is as guaranteed as the IR/RF remote. I was really hoping CEC would be the answer. 😢

Minims, 2024-03-18:
Same here. I need to power on / power off one time via IR, to be able to control via CEC the next time.

gekkie, 2024-04-20:
Not going to work unfortunatly. When the TV is OFF OFF then only some systems are active. Looks like only IR is on.

ntzb, 2024-05-03:
@Minims @iDVB did you try using 5V on pin 18?

Minims, 2024-05-03:
@ntzb I have wired the 3 cables, GND, CEC & 5v. All is working fine.

Minims, 2024-05-03:
@ntzb my current wiring is : 

* 17 -> (GND)
* 18 -> (5V)
* 13 -> (GPIO4)

I have a regulator 5v > 3v3 for the ESP, but i use the USB 5v for pin 18. 
No issue since month on my 15 years old TV. 
The hdmi cable is connected on an HDMI switch. 
But for debug puprose it was connected directly to TV without issue too.

closed 2024-07-16 by github-actions[bot]