Palakis/esphome-native-hdmi-cec #6: Implement ID

gekkie, 2024-04-20:
Hi,

First off LOVE THIS! after 2 months of screwing around with other options this one worked for me and saved my project!!!

Would it be possible to implement id in esphome. The reason is that I would love to use 3 HDMI CEC outputs.

Let me know hwo to buy you a coffee(tm) for all the hard work.

Palakis, 2024-05-12:
Hi @gekkie! Thank you for your kind words. I'll take care of this.

Palakis, 2024-05-12:
Thanks for the kind words @gekkie! I'll look into this.

Palakis, 2024-05-12:
@gekkie 

I whipped up something on a separate `feature/multiple-buses` branch. Can you please give it a try?

Update your `external_components` section so that `source` points to the right branch:
```yaml
external_components:
  - source: github://Palakis/esphome-hdmi-cec@feature/multiple-buses
```

And then declare several CEC buses like so:
```yaml
hdmi_cec:
  - id: cec1
    pin: GPIO26
    address: 0x5
    physical_address: 0x4000
    
  - id: cec2
    pin: GPIO27
    address: 0x6
    physical_address: 0x4000
```

solarmill, 2024-06-02:
Were you planning to use it as some sort of "bridge" to communicate with two different CEC networks?

gekkie, 2024-06-02:
Nope. The goal is to have a bt remote connect to an esphome, then according to which hdmi system is selected (via a modded hdmi switcher, added esphome to that) send vol up to either cec1 or cec2That’s the plan. Next week i start coding and will let you know if it worksCheersGeordy Korte

solarmill, 2024-08-07:
not stale.

solarmill, 2024-08-14:
@gekkie did you get this working?

gekkie, 2024-08-15:
Hi,

Not yet to be honest. Just finished the installation and ran into a bunch
of issues which set me back. The biggest is that when I use the CEC code it
causes the Stepper motors to jitter and hang. Before you go off and start
thinking about this, it is most probably an issue on my side. Instead of
using separate wire looms for each of my components (fans, steppers,
switches, displays etc) I decided to aggregate everything into one big
cable (30 strands) and a big beffy connector... Thought the idea was great
until I ran into crosstalk on the wires and most of my endstop switches
were firing randomly.

Like I said this is going to take some time, so I would close this ticket
instead of waiting for me to finish this simple project that is turning
into "the project from hell".

J4nsen, 2024-09-20:
I could imagine another use case for this. In my home theatre I have a projector which does not support HDMI CEC and a HTPC which also does not support it. However, my AV receiver does and it has a nice remote.

At the moment I'm using a Pulse-Eight adapter for my HTPC and with this project [I emulate a running TV](https://github.com/Pulse-Eight/libcec/issues/657) to receive key presses of the remote.

Please correct me if I'm wrong, but shouldn't it be possible to emulate a TV with `id: tv` and a playback device with `id: kodi` to receive the keypresses and send them via mqtt to kodi?

gekkie, 2024-09-24:
Hi Jan,

Kind of off -topic but what the heck. CEC-O-MATIC.COM is your go to place
to get all the commands you are going to need. In principle what you are
saying is correct and will work without to much hassel. This is what I
would do. Setup hdmi-cec on an ESPhome thinging (you choose) and set it to
listen broadcast mode. Because CEC is a 1980 protocol it's about as safe as
an open door and all commands on the CEC bus will be received on the ESP.
In the YAML you can setup what the ESP needs todo when it receives commands
which you can either push toward MQTT or directly to Kodi. If you need help
then feel free to contact me off list and I will try to help out where
needed or setup a reddit thread on a sub and ping me (Gek_kie)

closed 2024-12-01 by github-actions[bot]