Palakis/esphome-native-hdmi-cec #9: Replacing a Pi Zero W with cec-mqtt-bridge with an ESP32 running esphome-native-hdmi-cec

DIYtechie, 2024-07-03:
First of all thank you! This is exactly what I have been looking for. Low cost and easy to set up.

I have been using a Pi Zero W with [cec-mqtt-bridge](https://github.com/michaelarnauts/cec-mqtt-bridge) for 7-8 years and that also worked great, but it requires a Pi, an SD card, installing an OS and installing lots on top of that, which isn't ideal. Furthermore, the dependencies recently updated and made it dicfficult to deploy a new Pi zero in my setup. However, the cec-mqtt-bridge had 3 advantages that would be relevant to have withj esphome-native-hdmi-cec:

1) read all CEC traffic / incoming messages in Home Assistant and then decide which commands were relevant to act on in an automation. Is there an easy way to get the same information, e.g. written to a sensor with esphome-native-hdmi-cec? (Readable in Home Assistant)
2) send CEC commands using the [CEC-o-matic](https://www.cec-o-matic.com/index.php) format/syntax (no need for separate source and destination fields and 0x before each command etc)
3) not having to manually figure out the physical adress of the device itself

Maybe the last one is already on the to do list and the second one is a convenience thing, but especially the first one is something I would like to achieve. Any idea if this is possible?

DIYtechie, 2024-07-04:
> * # 1 is possible in your own config. just enable promiscuous mode on the CEC component, setup an MQTT client and pipe all CEC messages to the MQTT
> 
Thanks for your response. Sounds about right, and I have enables promiscuous mode, but don't know how to pipe the CEC messages anywhere? It doesn't need to via MQTT, a sensor would be just as good if not better. Similar to the button press entities in the example, but with a regular sensor entity.

Any help or an example would be much appreciated.

Palakis, 2024-07-04:
> [...] but don't know how to pipe the CEC messages anywhere?

> Any help or an example would be much appreciated.

Sure! Something in that fashion:

```yaml
mqtt:
  broker: 'homeassistantaddress'
  username: 'mqtt_user'
  password: 'mqtt_password'
  discovery: false # if you only want your own MQTT topics

hdmi_cec:
  pin: GPIO26
  address: 0xE
  physical_address: 0x4000
  promiscuous_mode: true 
  on_message:
    # No source, destination or opcode set => catch all messages
    - then:
        mqtt.publish:
          topic: cec_messages
          payload: !lambda |-
            std::vector<uint8_t> full_frame;
            full_frame.push_back((source << 4) | (destination & 0xF));
            full_frame.insert(full_frame.end(), data.begin(), data.end());
            return hdmi_cec::bytes_to_string(full_frame);       
```

The payloads will be in the CEC-O-MATIC format.

ntzb, 2024-07-04:
just tried the above code, but can't compile, getting:
```
esp-c3-cec.yaml: In lambda function:
esp-c3-cec.yaml:106:24: error: 'bytes_to_string' is not a member of 'esphome::hdmi_cec'
```

DIYtechie, 2024-07-05:
Thank you very much! Will try it out.

DIYtechie, 2024-07-05:
It works! Thank you so much. Fingers crossed for #2 (sending commands in cec-o-matic format), but this effectively allows me to do all I need for all my automations and fully replace my cec-mqtt-bridge so I will close this "issue". 

Thanks again. Awesome!