Palakis/esphome-native-hdmi-cec #10: ESP32 C3 supermini becomes unavailable

DIYtechie, 2024-07-07:
Downsizing from a full-sized ESP32 dev board (which works great) to a ESP32 C3 supermini, I am running into issues with the ESP rebooting or at least becoming unavailable. It can _send_ a single CEC command once, but as soon as it _receives_ CEC commands, it becomes unavailable.

I used the template in the readme (same as on my ESP32 dev board) and connected 5v, ground and gpio. Tried GPIO 5, 6 and 10 but the result is the same. It becomes unavailable. 

If I _don't_ connect the HDMI cable to my receiver, the C3 does not become unavailable, when I send the "all off" command. Same is the case if it is connected and all other devices are off. But if other devices are on and I send the "all off" command, the C3 becomes unavailable, which is what leads me to think that it cannot handle the commands it receives.

[image]
[image]

Any help is greatly appreciated. 

(I am in the proces of designing a case for this tiny project and will be happy to share as soon as it is done. Really want to use a C3 board to keep it tiny.)

nagyrobi, 2024-07-08:
Try grabbing serial logs while this happens.

DIYtechie, 2024-07-08:
> Try grabbing serial logs while this happens.

Will do. Should I plug it into the server runnning ESPHome dashboard or is the log over wifi just as good?

nagyrobi, 2024-07-08:
Serial logs - means over physically connected serial port - not wifi

DIYtechie, 2024-07-08:
config:
`esphome:
  name: supermini4
  friendly_name: supermini4

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: arduino

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: [redacted]

ota:
  - platform: esphome
    password: [redacted]

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Supermini4 Fallback Hotspot"
    password: [redacted]

captive_portal:

external_components:
  - source: github://Palakis/esphome-hdmi-cec

hdmi_cec:
  # Pick a GPIO pin that can do both input AND output
  pin: GPIO10 # Required
  # The address can be anything you want. Use 0xF if you only want to listen to the bus and not act like a standard device
  address: 0xE # Required
  # Physical address of the device. In this case: 4.0.0.0 (HDMI4 on the TV)
  # DDC support is not yet implemented, so you'll have to set this manually.
  physical_address: 0x4800 # Required
  # The name that will we displayed in the list of devices on your TV/receiver
  osd_name: "HDMI bridge" # Optional. Defaults to "esphome"
  # By default, promiscuous mode is disabled, so the component only handles directly-address messages (matching
  # the address configured above) and broadcast messages. Enabling promiscuous mode will make the component
  # listen for all messages (both in logs and the on_message triggers)
  promiscuous_mode: true # Optional. Defaults to false
  # By default, monitor mode is disabled, so the component can send messages and acknowledge incoming messages.
  # Enabling monitor mode lets the component act as a passive listener, disabling active manipulation of the CEC bus.
  monitor_mode: false # Optional. Defaults to false
  # List of triggers to handle specific commands. Each trigger has the following optional filter parameters:
  # - "source": match messages coming from the specified address
  # - "destination": match messages meant for the specified address
  # - "opcode": match messages bearing the specified opcode
  # - "data": exact-match on message content
  # Actions called from these triggers is called with "source", "destination" and "data" as parameters
  on_message:
    - opcode: 0x36 # opcode for "Standby"
      then:
        logger.log: "Got Standby command"

button:
  - platform: template
    name: "Turn all HDMI devices off"
    on_press:
      hdmi_cec.send:
        # "source" can optionally be set, like if you want to spoof another device's address
[... 288 lines cut ...]
[07:58:36][C][wifi:439]:   BSSID: [redacted]
[07:58:36][C][wifi:440]:   Hostname: 'supermini4'
[07:58:36][C][wifi:442]:   Signal strength: -22 dB ▂▄▆█
[07:58:36][C][wifi:446]:   Channel: 4
[07:58:36][C][wifi:447]:   Subnet: 255.255.255.0
[07:58:36][C][wifi:448]:   Gateway: 192.168.3.1
[07:58:36][C][wifi:449]:   DNS1: 192.168.3.1
[07:58:36][C][wifi:450]:   DNS2: 0.0.0.0
[07:58:36][C][logger:185]: Logger:
[07:58:36][C][logger:186]:   Level: DEBUG
[07:58:36][C][logger:188]:   Log Baud Rate: 115200
[07:58:36][C][logger:189]:   Hardware UART: USB_CDC
[07:58:36][C][captive_portal:088]: Captive Portal:
[07:58:36][C][mdns:115]: mDNS:
[07:58:36][C][mdns:116]:   Hostname: supermini4
[07:58:36][C][esphome.ota:073]: Over-The-Air updates:
[07:58:36][C][esphome.ota:074]:   Address: supermini4.local:3232
[07:58:36][C][esphome.ota:075]:   Version: 2
[07:58:36][C][esphome.ota:078]:   Password configured
[07:58:36][C][safe_mode:018]: Safe Mode:
[07:58:36][C][safe_mode:020]:   Boot considered successful after 60 seconds
[07:58:36][C][safe_mode:021]:   Invoke after 10 boot attempts
[07:58:36][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[07:58:36][W][safe_mode:031]: SAFE MODE IS ACTIVE
`

nagyrobi, 2024-07-08:
You should try asking in ESPHome discord in a thread what could be causing to trigger `esp_restart_noos_dig`.

Palakis, 2024-07-08:
Looks `gpio_intr_` is calling non-`IRAM_ATTR` functions, namely `vector::push_back` and `queue::push`, and that the ESP32-C3 is more strict about it than earlier models. I've ordered a couple of ESP32-C3 superminis to test and debug this on my end.

DIYtechie, 2024-07-08:
> Looks like something is wrong with the receive interrupt specifically on the ESP32-C3. I've ordered a couple of ESP32-C3 superminis to test and debug this on my end.

Awesome!

DIYtechie, 2024-08-11:
Any luck with getting the C3 supermini to work? :)

The case is coming along nicely:
[image]
[image]
[image]

TeDeVPrime, 2024-09-16:
these are nice! how and when can we get one? :)

DIYtechie, 2024-09-17:
> these are nice! how and when can we get one? :)

As soon as the C3 supermini is supported, I will make sure to share the 3D print file. As for the smallest HDMI female connector, I am using these: https://a.aliexpress.com/_EGrcIEt

TheAndi, 2024-10-15:
First of all: Great Work @Palakis and also @DIYtechie (I plan to design one that directly plugs in)! 

Any news on that? I have plenty of ESP32 C3 and would like to give it a try ;)

As my Android TV's API and WoL/WoWLAN is not that reliable, I hope this can be the solution. 

Thanks in advance :)

nagyrobi, 2024-10-15:
Try decreasing WiFi power on the C3 Supermini:
```yaml
wifi:
  output_power: 14dB
  enable_btm: true #optional
  enable_rrm: true #optional
```

DIYtechie, 2024-11-05:
> Looks like something is wrong with the receive interrupt specifically on the ESP32-C3. I've ordered a couple of ESP32-C3 superminis to test and debug this on my end.

Any luck getting this to work? Would be amazing to be able to make it as compact as possible.

titou4307, 2024-11-05:
Hello
How find the other template button ?

In the YAML there only the button "Turn everything off"

In video youtube of @DIYtechie  there is more buttons ???

DIYtechie, 2024-11-09:
> Hello How find the other template button ?
> 
> In the YAML there only the button "Turn everything off"
> 
> In video youtube of @DIYtechie there is more buttons ???

This is the wrong issue to ask that question. This thread is about C3 supermini compatibility issues.

However, If you open a new issue I will post the YAML for you and @Palakis can decide whether to include these examples in the readme.

titou4307, 2024-11-09:
@DIYtechie 

You have entirely reason....

I have open a new issue here : 

https://github.com/Palakis/esphome-native-hdmi-cec/issues/13

Thanks

DIYtechie, 2025-01-09:
> Looks like something is wrong with the receive interrupt specifically on the ESP32-C3. I've ordered a couple of ESP32-C3 superminis to test and debug this on my end.

Still have my fingers crossed, hoping that the debugging will lead to a solution. Would be truly amazing!

DIYtechie, 2025-04-22:
Hopefully this is not ruled out.

DIYtechie, 2025-04-22:
> Hi [@DIYtechie](https://github.com/DIYtechie), does the issue still happens with the latest version of this component? (Try busting your component cache with `refresh: 0s` under the repository url)

Based on https://github.com/Palakis/esphome-native-hdmi-cec/pull/16#issuecomment-2692799934 I assumed this would not be the case, but will give it a try in the weekend. Will be very exited if this works! Did you ever receive the ESP 32 S3 boards for testing?

DIYtechie, 2025-04-30:
> Hi [@DIYtechie](https://github.com/DIYtechie), does the issue still happens with the latest version of this component? (Try busting your component cache with `refresh: 0s` under the repository url)

Just tested (also using the refresh: 0s) and unfortunately the device becomes unavailable as soon as I have sent a command via CEC. Here is the log:

```
[23:25:29][C][safe_mode:020]:   Boot considered successful after 60 seconds
[23:25:29][C][safe_mode:021]:   Invoke after 10 boot attempts
[23:25:29][C][safe_mode:023]:   Remain in safe mode for 300 seconds
[23:25:30][C][api:140]: API Server:
[23:25:31][D][api:103]: Accepted 192.168.4.25
[23:25:31][D][api.connection:1801]: Home Assistant 2025.4.4 (192.168.4.25): Connected successfully
[23:25:44][D][button:010]: 'C3 Turn TV off' Pressed.
[23:25:44][D][hdmi_cec:203]: sending frame: 10:36
[23:25:44][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:25:44][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:25:44][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:25:44][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:25:44][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:25:44][E][hdmi_cec:235]: HDMICEC::send(): send failed after five attempts
[23:25:44][W][component:239]: Component api took a long time for an operation (80 ms).
[23:25:44][W][component:240]: Components should block for at most 30 ms.
[23:25:50][D][api:103]: Accepted 192.168.4.25
[23:25:50][W][api.connection:123]: : Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[23:26:27][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
[23:26:27][D][esp32.preferences:114]: Saving 1 preferences to flash...
[23:26:27][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[23:26:51][D][api:103]: Accepted 192.168.4.25
[23:26:51][W][api.connection:123]: : Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
[23:27:13][D][button:010]: 'C3 Turn TV off' Pressed.
[23:27:13][D][hdmi_cec:203]: sending frame: 10:36
[23:27:13][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:13][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:13][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:13][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:13][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:13][E][hdmi_cec:235]: HDMICEC::send(): send failed after five attempts
[23:27:13][W][component:239]: Component api took a long time for an operation (83 ms).
[23:27:13][W][component:240]: Components should block for at most 30 ms.
[23:27:23][D][button:010]: 'C3 Turn all HDMI devices off' Pressed.
[23:27:23][D][hdmi_cec:203]: sending frame: EF:36
[23:27:23][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:23][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:23][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:23][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:23][D][hdmi_cec:229]: HDMICEC::send(): frame not sent: Bus Collision
[23:27:23][E][hdmi_cec:235]: HDMICEC::send(): send failed after five attempts
[23:27:23][W][component:239]: Component api took a long time for an operation (82 ms).
[23:27:23][W][component:240]: Components should block for at most 30 ms.
[23:27:51][D][api:103]: Accepted 192.168.4.25
[23:27:51][W][api.connection:123]: : Socket operation failed: HANDSHAKESTATE_READ_FAILED errno=11
```

DIYtechie, 2025-05-07:
Could it be that the C3 requires explicit configuration to use half-duplex mode for the data, as the driver defaults to full-duplex unless specified? 

And could pin 10 on the C3 generally be a no go for UART (if is is part of the USB_JTAG peripheral and may suffer from resource conflicts if I connect the C3 USB for power)? I tried different ones, but can't remember if I tried pin 20 or 21.

I asked ChatGPT and these were suggested sources of issues with HDMI CEC using the C3 board.

DIYtechie, 2025-05-07:
Unfortunately, using pin 21 did nothing and manually configuring UART half duplex is apparently not supported (or necessary?) with ESPHome. So still no luck.

DIYtechie, 2025-05-29:
I just tested with my ESP32 C3 it finally works! Amazing! I am using pin 10 for data with no issue.

When using the esp-idf my C3 does not become unavailable.

I think I tried using the esp-idf before, but I am not sure.

Thank you so much! Will be releasing the 3D print file for [the case I designed](https://github.com/Palakis/esphome-native-hdmi-cec/issues/10#issuecomment-2282854890) asap!

JosVanEijndhoven, 2025-05-29:
Hi @DIYtechie ,
I am afraid you were just an hour early. I made a significant further modification on request of Palakis, just after your succes message. Maybe you can try to build and run again?

DIYtechie, 2025-05-29:
> Hi [@DIYtechie](https://github.com/DIYtechie) , I am afraid you were just an hour early. I made a significant further modification on request of Palakis, just after your succes message. Maybe you can try to build and run again?

Happy to test the latest version, but don't want to lose my current entities that are now finally working. How do I ensure that I pull the latest code without having to change the config?

DIYtechie, 2025-05-30:
Sorry for being unclear. I know that these are internal changes, but I am not sure how to ensure that I pull these changes in and test them without setting up a completely new ESP. Will any installation including 
```
external_components:
  - source: github://Palakis/esphome-hdmi-cec
```
pull the latest from GitHub every time, even if it is the same or will it use a previously downloaded code? I recall Palakis mentioned something about a component cache somewhere.

By config I just mean the specific ESPHome config for this. I have added several [buttons](https://github.com/Palakis/esphome-native-hdmi-cec/issues/13#issuecomment-2466156017) and CEC [decoder sensors](https://github.com/Palakis/esphome-native-hdmi-cec/issues/14#issuecomment-2845890263) in addition to MQTT that I would like to be able to keep. Based on your response, I guess I don't need to change any of this.

JosVanEijndhoven, 2025-06-04:
Hi @DIYtechie ,
how did you test the code update that I made for resolving the crashing of the esp32c3?
For me that seems to work for the esp32c3, with both the`arduino` and the `esp-idf` frameworks.

My prepared update is still an open 'pull request' ([#20](https://github.com/Palakis/esphome-native-hdmi-cec/pull/20)). So you do not yet obtain that automatically.
The [syntax for specifying an external component ](https://esphome.io/components/external_components.html) does not specify how to write such. As far as I know, you can test my update by specifying the branch name of my repository fork of Palakis'.
That would be something like::
```
   - source: github://JosVanEijndhoven/esphome-native-hdmi-cec@fix-isr-esp32c3`
     components: [hdmi_cec]
```
Did you use that url for your recent application test?

JosVanEijndhoven, 2025-06-05:
Yes, you would just need to change the `github:` url to test this update.

DIYtechie, 2025-06-05:
Sidenote:

@Palakis and @JosVanEijndhoven I just released a case for the ESP32 C3: https://makerworld.com/en/models/1488957-esp32-c3-hdmi-case#profileId-1556002

I am so excited and grateful that this firmware exists and that it works on the C3 so if you don't have a 3D printer and would like a case, just send me your address and I will be happy to ship it to you.

DIYtechie, 2025-06-05:
> Yes, you would just need to change the `github:` url to test this update.

Hmm. Did not seem to work:

```
INFO ESPHome 2025.5.2
INFO Reading configuration /config/esphome/hdmi.yaml...
INFO Updating https://github.com/JosVanEijndhoven/esphome-native-hdmi-cec.git@fix-isr-esp32c3
INFO Generating C++ source...
INFO Updating https://github.com/espressif/esp-protocols.git@mdns-v1.8.2
INFO Compiling app...
Processing hdmi-cec-bridge (board: esp32-c3-devkitm-1; framework: espidf; platform: https://github.com/pioarduino/platform-espressif32.git#51.03.07)
--------------------------------------------------------------------------------
HARDWARE: ESP32C3 160MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.50106.0 (5.1.6) 
 - tool-cmake @ 3.21.3 
 - tool-esptoolpy @ 4.8.1 
 - tool-mklittlefs @ 3.2.0 
 - tool-ninja @ 1.7.1 
 - tool-riscv32-esp-elf-gdb @ 12.1.0+20221002 
 - tool-xtensa-esp-elf-gdb @ 12.1.0+20221002 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-riscv32-esp @ 12.2.0+20230208
Reading CMake configuration...
Dependency Graph
|-- ArduinoJson @ 6.18.5
|-- noise-c @ 0.1.6
Compiling .pioenvs/hdmi-cec-bridge/src/main.cpp.o
/config/esphome/hdmi.yaml: In lambda function:
/config/esphome/hdmi.yaml:94:40: error: cannot convert 'std::vector<unsigned char>' to 'const esphome::hdmi_cec::Frame*' {aka 'const std::vector<unsigned char>*'}
   94 |             return hdmi_cec::bytes_to_string(full_frame);
      |                                        ^~~~~~~~~~
      |                                        |
      |                                        std::vector<unsigned char>
In file included from src/esphome.h:23,
                 from src/main.cpp:3:
src/esphome/components/hdmi_cec/hdmi_cec.h:16:42: note:   initializing argument 1 of 'std::string esphome::hdmi_cec::bytes_to_string(const Frame*)'
   16 | std::string bytes_to_string(const Frame *bytes);
      |                             ~~~~~~~~~~~~~^~~~~
*** [.pioenvs/hdmi-cec-bridge/src/main.cpp.o] Error 1
========================== [FAILED] Took 5.04 seconds ==========================
```
Will probably have to wait and try it when it is merged.

cap60552, 2025-06-09:
I was having the constant reboot issue as well. 
Switching to "- source: github://JosVanEijndhoven/esphome-native-hdmi-cec@fix-isr-esp32c3"  worked great and I'm finally able to get some things to work.  ( I can at least power on and off the TV,  having trouble switching HDMI inputs, but I don't think that's a problem with the library.  )

closed 2025-06-15 by Palakis