Palakis/esphome-native-hdmi-cec #12: [Bug]: Timing issues with sent frames?

pzich, 2024-11-04:
### The problem

I'm looking to place my HDMI-CEC adapter (in this case, a Pulse-Eight [USB-CEC Adapter](https://www.pulse-eight.com/p/104/usb-hdmi-cec-adapter)), and this component is seeming like a great option! Thank you for building this!

However, while debugging things, I've been looking at the log output from the Pulse-Eight device, and every time the ESP sends a frame, there are a ton of `HIGH_ERROR` messages, suggesting there is something wrong with the timing of these messages?

[image]

I'm far from an expert when it comes to precise timing with MCUs, but looking at the code, I was wondering if a few things could be the culprit:
- There seems to be a mix of `delayMicroseconds()` and `delay_microseconds_safe()`. Would it be better to use only one of these for all cases?
- I noticed that `send_and_read_ack_()` is structured using an initial `start_us = micros()`, and after using `delayMicroseconds(HIGH_BIT_US)`, the timing is controlled by checking against the start time with `while((micros() - start_us) < SAFE_SAMPLE_US)` and `while((micros() - start_us) < TOTAL_BIT_US)`, while `send_bit_()` just uses `pin_->digital_write()`s and `delayMicroseconds()`. I'd imagine the timing for these is pretty reliable, but perhaps it would also benefit from a similar looped delay structure?
- As far as I can tell, the `InterruptLock` in `send_frame_()` is all that's needed to prevent interrupts, but perhaps something is still causing an interrupt or other unexpected delay?

### Which version of ESPHome was used when this happened?

2024.10.2

### What platform are you using?

ESP32-IDF

### Board

ESP32-DevKitC

### Example YAML snippet

```yaml
Just the plain `hdmi_cec` setup will trigger this when it responds (e.g. responding to `0x83`):

hdmi_cec:
  pin: GPIO13
  address: 0x8
  physical_address: 0x3000

```

Also happens for `hdmi_cec.send` actions:
```
  - platform: template
    name: "HDMI3"
    on_press:
      hdmi_cec.send:
        destination: 0xF
        data: [0x82, 0x30, 0x00]
```
```

### Anything in the logs that might be useful?

```txt
There seem to be `Component hdmi_cec took a long time for an operation` message (almost?) every time it sends. Nothing else out of the ordinary in the logs

[15:08:04][D][hdmi_cec:076]: frame received: 08:8C
[15:08:04][D][hdmi_cec:205]: sending frame: 80:00:8C:00
[15:08:04][D][hdmi_cec:219]: HDMICEC::send(): frame sent and acknowledged
[15:08:04][W][component:237]: Component hdmi_cec took a long time for an operation (121 ms).
[15:08:04][W][component:238]: Components should block for at most 30 ms.
```
```

TheAndi, 2024-11-09:
I can confirm that something is off with this implementation!

I am **not** able to control my Philips OLED808 with this. The TV can send me requests, but even if I reply to them, the TV keeps asking, this does not happen with [ceclib ](https://github.com/Pulse-Eight/libcec/tree/master) on an Orange Pi. 

Furthermore, the TV also ignores all my commands or requests. The **only** thing that triggers the TV is XF:36 (OFF as broadcast)

On the exact same Hardware (ESP32 and ESP32C3) I tested [Tasmota HDMI CEC](https://tasmota.github.io/docs/HDMI_CEC/) as well as the older [ESPHome CEC component](https://github.com/johnboiles/esphome-hdmi-cec) on which this is based on.

Looking forward to a fix. I can also beta-test new code if required ;)

pzich, 2025-01-09:
I tried a fresh build with the latest ESPHome but am still seeing timing issues.

Palakis, 2025-02-20:
@pzich 

> There seems to be a mix of delayMicroseconds() and delay_microseconds_safe(). Would it be better to use only one of these for all cases?

You're correct. I have replaced all calls to `delayMicroseconds()` with `delay_microseconds_safe()`.

> I noticed that send_and_read_ack_() is structured using an initial start_us = micros(), and after using delayMicroseconds(HIGH_BIT_US), the timing is controlled by checking against the start time [...]

I've replaced the flaky busy-wait loops with calls to `delay_microseconds_safe()`. Should make the delay work nicer with the WDT and WiFi/BT tasks.

> As far as I can tell, the InterruptLock in send_frame_() is all that's needed to prevent interrupts, but perhaps something is still causing an interrupt or other unexpected delay?

Let's first see if the two fixes above improve the timing situation. Both of them have already been committed to the `main` branch.

JosVanEijndhoven, 2025-02-27:
I am using this repro to create an HDMI 'audio device' with ARC support. This requires quite a lot of in- and out-bound messages. It seems to work fine, but sometimes (like every hour or so) there is a timing error related to sent messages. To me it seems that the `InterruptLock` in `send_frame_()` keeps its lock for too long, causing the OS to crash and reboot. See a section of a log-file below.  A very good job was done in this repo to implement the receiving messages all interrupt-driven.  Maybe the sending of messages should be implemented with a similar interrupt scheme? I can volunteer in helping to do so.

In the esphome log I see frequent messages on 'took a long time':

```
[13:32:39 [hdmi_cec:20 <UNKNOWN>3]: sending frame: 50:C0 <start ARC>
[13:32:39 [D][hdmi_cec:22 <UNKNOWN>4]: HDMICEC::send(): frame not sent: No Ack received
[13:32:39 [D][hdmi_cec:22 <UNKNOWN>0]: HDMICEC::send(): frame sent and acknowledged
[13:32:39 [W][component:237]: Component hdmi_cec took a long time for an operation (189 ms).
[13:32:39 [W][component:238]: Components should block for at most 30 ms.
[13:32:39 [D][hdmi_cec:07 <tuner device status>5]: frame received: 05:C3 <request ARC start>
```

And worse, sometimes a full system reset from the 'watchdog':

```
[13:32:40 [D][hdmi_cec:07 <tuner device status>5]: frame received: 05:C1 <report ARC started>
[13:32:45 E (286538) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
[13:32:45 E (286538) task_wdt:  - loopTask (CPU 1)
[13:32:45 E (286538) task_wdt: Tasks currently running:
[13:32:45 E (286538) task_wdt: CPU 0: IDLE
[13:32:45 E (286538) task_wdt: CPU 1: IDLE
[13:32:45 E (286538) task_wdt: Aborting.
[13:32:45 ESP-ROM:esp32s3-20210327
[13:32:45 Build:Mar 27 2021
[13:32:45 rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
[13:32:45 Saved PC:0x40378010
WARNING Decoded 0x40378010: esp_restart_noos at /Users/ficeto/Desktop/ESP32/ESP32S2/esp-idf-public/components/esp_system/port/soc/esp32s3/system_internal.c:143 (discriminator 1)
```

JosVanEijndhoven, 2025-03-02:
Hi @pzich , I have proposed a code update to fix #15  . Although I originally thought that to be a different issue, and I don't know about the adapter you use, it might also help on your #12. It could be useful to test whether it helps, even before my branch would be merged in? 

To do such test, you would need to modify your project yaml where you specify the `external_components`, to pick up the version in my branch:
 ```
 - source:
     type: git
     url: https://github.com/JosVanEijndhoven/esphome-native-hdmi-cec
     ref: safe_collision
   components: [hdmi_cec]
```
Clearly, this git specification would be temporary, until (some reviewed/updated) version would be integrated in the original main repo.

pzich, 2025-05-02:
Hoping to give this another test soon!

pzich, 2025-05-08:
Overall this seems to be working _much_ better than before, but I am still sometimes seeing reports of timing issues. 🤔

[image]

Not too sure how best to debug. In particular this was it responding to a "give device vendor ID" request from the TV.

benooye, 2025-05-11:
Dear @JosVanEijndhoven you mentioned earlier to try to implement an HDMI 'audio device' with ARC support
I'm trying also without success to build a fake ARC device on my Samsung S90C : the TV switches properly for 10 seconds on the fake ARC device and then switches back the TV speakers.
Maybe it could be linked to SHORT_AUDIO_DESCRIPTOR issue or the timing issues described here?, I'm experimenting...
Did you succeeded on your side? 
Would you mind to share your ESPHome code?
Thanks a lot in advance!

JosVanEijndhoven, 2025-05-12:
Hi @benooye , yes, my ARC implementation is working fine as of recently. As you say in your update: I also found that the request-answer protocol can be critical, and sometimes only weakly described. Besides protocol issues, I did also have timing problems: the hdmi_cec component in this repo creates long periods of blocking all interrupts, occasionally causing failures in the underlying OS (probably related with wifi), resulting in unexpected reboots. Such reboots caused my ARC to drop and renegotiate. I have been working on a further refinement of this component to have two improvements: a) avoid this long interrupt blocking, b) optionally utilize a UART to avoid long cpu-bound tasks and the related OS warnings of "took a long time for an operation". I intend to add the updated component to the esphome repo itself, but that might take some more time. If you want to, you can try my current/early version by taking the 'external component' from [my fork](https://github.com/JosVanEijndhoven/esphome).

The yaml of my audio ARC application is quite complex as it adds low-level control of my home-design DIY DAC. Also, I added the CEC power up/down protocols and the HDMI 'hpd' (hot plug detect) as sensor input. But you are free to [take look at it](https://gitlab.com/JosVanEijndhoven/dacesphome/-/blob/hdmi-cec/dac.yaml?ref_type=heads).

benooye, 2025-05-14:
> Hi [@benooye](https://github.com/benooye) , yes, my ARC implementation is working fine as of recently. As you say in your update: I also found that the request-answer protocol can be critical, and [...]

> Hi [@benooye](https://github.com/benooye) , yes, my ARC implementation is working fine as of recently. As you say in your update: I also found that the request-answer protocol can be critical, and [...]

Thanks a lot for your answer!
I will check what you performed with great interest 
Thanks again

closed 2025-07-21 by github-actions[bot]

ntzb, 2026-09-06:
This might be what you were circling - the micros-based waits in `send_and_read_ack_` (now `send_high_and_test_`), and interrupts landing mid-bit.

Those waits are computed as `CONSTANT - (micros() - start)`, unsigned, so once the bit has overrun the subtraction underflows. On a dual-core board I'd expect that to show as the stretched bit and the `HIGH_ERROR` you saw without necessarily crashing; on a single core it also stalls the loop task.

#56 clamps both waits, if it's worth a retest.