Palakis/esphome-native-hdmi-cec #14: Updated code to include CEC message decoder

gekkie, 2025-02-19:
Hey Palakis,

Could you reach out to me at gkorte [] gmail? I have updated the code to decode CEC messages and have no clue on how to get them into your code, but feel that it would be a nice add on.

Palakis, 2025-02-20:
Hi @gekkie, could you please go into a bit more detail into what you've done and want to achieve, potentially including examples please?

gekkie, 2025-02-20:
Hi,

This is what I have working now:
_hdmi_cec:
  pin: ${HDMI_CEC}
  address: 0xF #Broadcast only
  physical_address: 0x1000 # HDMI1 port (I think)
  promiscuous_mode: true
  monitor_mode: true
  decode mode: true`_

`
[19:00:38][D][hdmi_cec:290]: frame received: 50:83                          	 Audio -> TV: Give physical adres
[19:00:39][D][hdmi_cec:290]: frame received: 0F:84:00:00:00                 	 TV -> Bcast: Report physical adres
[19:00:39][D][hdmi_cec:290]: frame received: 5F:87:00:E0:36                 	 Audio -> Bcast: Device vendor ID
[19:00:39][D][hdmi_cec:290]: frame received: 5F:84:30:00:05                 	 Audio -> Bcast: Report physical adres
[19:00:39][D][hdmi_cec:290]: frame received: 50:46                          	 Audio -> TV: Give OSD Name
[19:00:39][D][hdmi_cec:290]: frame received: 05:47:41:4D:4C:5F:54:56        	 TV -> Audio: Set OSD name
[19:00:39][D][hdmi_cec:290]: frame received: 58:46                          	 Audio -> Playback 2: Give OSD Name
[19:00:39][D][hdmi_cec:290]: frame received: 85:47:53:48:49:45:4C:44        	 Playback 2 -> Audio: Set OSD name
[19:00:40][D][hdmi_cec:290]: frame received: 5F:A0:00:E0:36:D8              	 Audio -> Bcast: Vendor command with ID
[19:00:40][D][hdmi_cec:290]: frame received: 5F:A0:00:E0:36:D5              	 Audio -> Bcast: Vendor command with ID
[19:00:40][D][hdmi_cec:290]: frame received: 50:8F                          	 Audio -> TV: Give device power status
[19:00:40][D][hdmi_cec:290]: frame received: 05:90:01                       	 TV -> Audio: Report power status
[19:00:40][D][hdmi_cec:290]: frame received: 50:C0                          	 Audio -> TV: Unknown CEC Command 
`

pbo10, 2025-02-21:
This would be a really nice addition for me, I'd like to see this added if at all possible.

closed 2025-05-01 by github-actions[bot]

DIYtechie, 2025-05-01:
Finally figured it out! Hopefully this can be shared in the code or at least as an example in the readme.

This creates two text sensors with raw and translated commands (remember to exclude the entities from logbook and recorder to save space):

```
hdmi_cec:
  # Pick a GPIO pin that can do both input AND output
  pin: GPIO26 # Required
  # The address can be anything you want. Use 0xF if you only want to listen to the bus and not act like a standard device
  address: 0xE # Required
  # Physical address of the device. In this case: 4.0.0.0 (HDMI4 on the TV)
  # DDC support is not yet implemented, so you'll have to set this manually.
  physical_address: 0x4200 # Required
  # The name that will we displayed in the list of devices on your TV/receiver
  osd_name: "HDMI bridge" # Optional. Defaults to "esphome"
  # By default, promiscuous mode is disabled, so the component only handles directly-address messages (matching
  # the address configured above) and broadcast messages. Enabling promiscuous mode will make the component
  # listen for all messages (both in logs and the on_message triggers)
  promiscuous_mode: true # Optional. Defaults to false
  # By default, monitor mode is disabled, so the component can send messages and acknowledge incoming messages.
  # Enabling monitor mode lets the component act as a passive listener, disabling active manipulation of the CEC bus.
  monitor_mode: false # Optional. Defaults to false
  # List of triggers to handle specific commands. Each trigger has the following optional filter parameters:
  # - "source": match messages coming from the specified address
  # - "destination": match messages meant for the specified address
  # - "opcode": match messages bearing the specified opcode
  # - "data": exact-match on message content
  # Actions called from these triggers is called with "source", "destination" and "data" as parameters
  on_message:
  - then:
      - lambda: |-
          // — Build raw frame string —
          std::vector<uint8_t> frame;
          frame.push_back((source << 4) | (destination & 0xF));
          frame.insert(frame.end(), data.begin(), data.end());
          std::string raw = hdmi_cec::bytes_to_string(frame);
          id(cec_raw_message).publish_state(raw);

          // — Helper lambdas —
          auto formatWithPeriods = [&](uint16_t num) {
            std::string s = std::to_string(num), out;
            for (char c : s) { out += c; out += '.'; }
            if (!out.empty()) out.pop_back();
            return out;
          };
          auto toHex = [&](uint8_t b) {
            char buf[3]; sprintf(buf, "%02X", b);
            return std::string(buf);
          };
          auto getDeviceName = [&](uint8_t nib) {
            static const char* names[] = {
              "TV","Recorder 1","Recorder 2","Tuner 1","Playback 1",
              "Audio system","Tuner 2","Tuner 3","Playback 2","Recorder 3",
              "Tuner 4","Playback 3","Reserved 1","Reserved 2","Free Use","Broadcast"
            };
            return std::string(names[nib & 0xF]);
          };
          // apply your custom names
          auto applyCustom = [&](const std::string &n) {
            if (n == "Playback 1")       return std::string("Apple TV");
            if (n == "Playback 2")       return std::string("PlayStation");
            if (n == "Audio system")     return std::string("Denon AVR");
            return n;
          };

          // — Build device prefix —
          uint8_t first = frame[0], src = first >> 4, dst = first & 0xF;
          std::string dev_src = applyCustom(getDeviceName(src));
          std::string dev_dst = applyCustom(getDeviceName(dst));
          std::string prefix = dev_src
[... 114 lines cut ...]
                  : b2 == 0x02 ? " (In transition from Standby to On)"
                  : b2 == 0x03 ? " (Power Off)"
                  : " (Unknown: " + toHex(b2) + ")");
                break;
              default:
                action = "Unknown action (0x" + toHex(opcode) + ")";
                break;
            }

          // — Publish the human-readable translation —
          std::string translated = prefix + " " + action;
          id(cec_translated_message).publish_state(translated);

text_sensor:
  - platform: template
    name: "HDMI CEC Raw Message"
    id: cec_raw_message
    update_interval: never

  - platform: template
    name: "HDMI CEC Translated Message"
    id: cec_translated_message
    update_interval: never

```

Minims, 2025-05-04:
@DIYtechie Is there a way to translate directly the name announce by the device instead of mapping custom names ?

DIYtechie, 2025-05-04:
> [@DIYtechie](https://github.com/DIYtechie) Is there a way to translate directly the name announce by the device instead of mapping custom names ?

Yes, I’m pretty sure you can request device/brand names and physical address of each device via HDMI CEC and then use that, but this is bit more complicated. If anyone else can add this, thats great.

Minims, 2025-06-06:
> Yes, I’m pretty sure you can request device/brand names and physical address of each device via HDMI CEC and then use that, but this is bit more complicated. If anyone else can add this, that would be great!

This seems working, i can detect TV, Playstation & Freebox

````
  on_message:
    - then:
        - lambda: |-
            // — Build raw frame string —
            std::vector<uint8_t> frame;
            frame.push_back((source << 4) | (destination & 0xF));
            frame.insert(frame.end(), data.begin(), data.end());
            std::string raw = hdmi_cec::bytes_to_string(frame);
            id(cec_raw_message).publish_state(raw);

            // — Helper lambdas —
            auto formatWithPeriods = [&](uint16_t num) {
              std::string s = std::to_string(num), out;
              for (char c : s) { out += c; out += '.'; }
              if (!out.empty()) out.pop_back();
              return out;
            };
            auto toHex = [&](uint8_t b) {
              char buf[3]; sprintf(buf, "%02X", b);
              return std::string(buf);
            };
            auto getDeviceName = [&](uint8_t nib) {
              static const char* names[] = {
                "TV","Recorder 1","Recorder 2","Tuner 1","Playback 1",
                "Audio system","Tuner 2","Tuner 3","Playback 2","Recorder 3",
                "Tuner 4","Playback 3","Reserved 1","Reserved 2","Free Use","Broadcast"
              };
              return std::string(names[nib & 0xF]);
            };
            // apply your custom names
            auto applyCustom = [&](const std::string &n) {
              if (n == "Playback 1")       return std::string("Playback 1");
              if (n == "Playback 2")       return std::string("Playback 2");
              if (n == "Playback 3")       return std::string("Playback 3");
              if (n == "Audio system")     return std::string("Denon AVR");
              return n;
            };

            // — Dynamic device name mapping —
            static std::map<uint8_t, std::string> dynamic_names;

            // — Build device prefix —
            uint8_t first = frame[0], src = first >> 4, dst = first & 0xF;
            auto getDynamicName = [&](uint8_t nib) -> std::string {
              auto it = dynamic_names.find(nib);
              if (it != dynamic_names.end() && !it->second.empty())
                return it->second;
              return applyCustom(getDeviceName(nib));
            };
            std::string dev_src = getDynamicName(src);
            std::string dev_dst = getDynamicName(dst);
            std::string prefix = dev_src
              + (dst == 0xF
                  ? " broadcasting:"
                  : " to " + dev_dst + ":");

            // — Parse opcode & args —
            uint8_t opcode = frame.size()>1 ? frame[1] : 0x00;
            uint8_t b2     = frame.size()>2 ? frame[2] : 0x00;
            uint8_t b3     = frame.size()>3 ? frame[3] : 0x00;
            uint16_t phys  = (b2 << 8) | b3;
            int volume     = static_cast<int>(b2) - 2;
            std::string action;

            // — Full translation switch —
[... 110 lines cut ...]
                  for (size_t i = 2; i < frame.size(); ++i) {
                    if (i > 2) data_str += " ";
                    data_str += toHex(frame[i]);
                  }
                  action = "Vendor command with data: [" + data_str + "]";
                  break;
                }
                case 0x90:
                  action = "Reporting power status"
                    + (b2 == 0x00 ? " (Power On)"
                    : b2 == 0x01 ? " (Standby)"
                    : b2 == 0x02 ? " (Waking up)"
                    : b2 == 0x03 ? " (Power Off)"
                    : " (Unknown: " + toHex(b2) + ")");
                  break;
                default:
                  action = "Unknown action (0x" + toHex(opcode) + ")";
                  break;
              }

            // — Publish the human-readable translation —
            std::string translated = prefix + " " + action;
            id(cec_translated_message).publish_state(translated);

```

DIYtechie, 2025-06-07:
> This seems working, i can detect TV, Playstation & Freebox

I just tested it and automatically fetching the names works! So awesome! Thanks! 

Tested for a few hours and a couple of times the full OSD name was not received by my ESP32 C3 (PlayStation 4 was Pn0, then PlayStatp, AVR-3200W was AVR? and Apple TV was Appv). Now sure why. However, the correct full names were later received and utilized.

I think custom names are still relevant to ensure you can use translated CEC messages in automations. Alternatively, raw messages can be used.

I have added a few more improvements below:
- fixed parse mistake for physical address
- custom names only applied when defined (and take precedence if they are)
- removed nonsensical HDMI version appendix (still available as raw message of course)
- French comments translated into English

```
    - then:
        - lambda: |-
            // — Build raw frame string —
            std::vector<uint8_t> frame;
            frame.push_back((source << 4) | (destination & 0xF));
            frame.insert(frame.end(), data.begin(), data.end());
            std::string raw = hdmi_cec::bytes_to_string(frame);
            id(cec_raw_message).publish_state(raw);

            // — Helper functions —
            auto toHex = [&](uint8_t b) {
              char buf[3]; sprintf(buf, "%02X", b);
              return std::string(buf);
            };

            auto getDeviceName = [&](uint8_t nib) {
              static const char* names[] = {
                "TV","Recorder 1","Recorder 2","Tuner 1","Playback 1",
                "Audio system","Tuner 2","Tuner 3","Playback 2","Recorder 3",
                "Tuner 4","Playback 3","Reserved 1","Reserved 2","Free Use","Broadcast"
              };
              return std::string(names[nib & 0xF]);
            };

            // — Custom name overrides (by HDMI logical address) —
            auto applyCustom = [&](uint8_t addr, const std::string &fallback) -> std::string {
              // Uncomment relevant lines to override device names:

              // === TV ===
              // if (addr == 0x0) return "Living Room TV";         // TV

              // === Audio ===
              // if (addr == 0x5) return "Denon AVR";              // Audio system

              // === Playback devices ===
              // if (addr == 0x4) return "Apple TV";               // Playback 1
              // if (addr == 0x8) return "Chromecast";             // Playback 2
              // if (addr == 0xB) return "Blu-ray Player";         // Playback 3

              // === Recorders ===
              // if (addr == 0x1) return "HD PVR";                 // Recorder 1
              // if (addr == 0x2) return "DVD Recorder";           // Recorder 2
              // if (addr == 0x9) return "Spare Recorder";         // Recorder 3

              // === Tuners ===
              // if (addr == 0x3) return "Set-top Box";            // Tuner 1
              // if (addr == 0x6) return "FM Radio";               // Tuner 2
              // if (addr == 0x7) return "DAB Receiver";           // Tuner 3
              // if (addr == 0xA) return "Extra Tuner";            // Tuner 4

              // === Reserved ===
              // if (addr == 0xC) return "Custom 13";              // Reserved 1
              // if (addr == 0xD) return "Custom 14";              // Reserved 2

              // === Other ===
[... 158 lines cut ...]
                    : b2 == 0x01 ? " (Standby)"
                    : b2 == 0x02 ? " (Waking up)"
                    : b2 == 0x03 ? " (Power Off)"
                    : " (Unknown: " + toHex(b2) + ")");
                  break;
                default:
                  action = "Unknown action (" + raw + ")";
                  break;
            }

            // — Publish final translation —
            std::string translated = prefix + " " + action;
            id(cec_translated_message).publish_state(translated);

text_sensor:
  - platform: template
    name: "HDMI CEC Raw Message"
    id: cec_raw_message
    update_interval: never

  - platform: template
    name: "HDMI CEC Translated Message"
    id: cec_translated_message
    update_interval: never
```

DIYtechie, 2025-06-07:
Hmm... Sometimes the correct OSD names are first correctly received and utilized and then later transmitted but not received at all (which then defaults to the standard name)

As can be seen below Playback 1 is first correctly translated to Apple TV based on OSB, but then later receives empty OSD and defaults back to Playback 1.

Playback 2 is first correctly translated to PlayStation 4 based on OSB, but then later receives empty OSD and defaults back to Playback 2.

```
[13:38:49][D][text_sensor:064]: 'HDMI CEC Raw Message': Sending state '40:47:41:70:70:6C:65:20:54:56'
[13:38:49][D][text_sensor:064]: 'HDMI CEC Translated Message': Sending state 'Playback 1 to TV: Set OSD display name: "Apple TV"'
[13:38:52][D][text_sensor:064]: 'HDMI CEC Raw Message': Sending state '40:47'
[13:38:52][D][text_sensor:064]: 'HDMI CEC Translated Message': Sending state 'Apple TV to TV: Set OSD display name: ""'
[13:39:10][D][text_sensor:064]: 'HDMI CEC Translated Message': Sending state 'Playback 2 to TV: Set OSD display name: "PlayStation 4"'
[13:39:10][D][hdmi_cec:089]: frame received: 08:A0:00:00:F0:23
[13:39:10][D][text_sensor:064]: 'HDMI CEC Raw Message': Sending state '08:A0:00:00:F0:23'
[13:39:10][D][text_sensor:064]: 'HDMI CEC Translated Message': Sending state 'TV to PlayStation 4: Routing information'
[13:40:20][D][text_sensor:064]: 'HDMI CEC Raw Message': Sending state '80:47'
[13:40:20][D][text_sensor:064]: 'HDMI CEC Translated Message': Sending state 'PlayStation 4 to TV: Set OSD display name: ""'
[13:40:20][D][text_sensor:064]: 'HDMI CEC Raw Message': Sending state '08:90:00'
[13:40:20][D][text_sensor:064]: 'HDMI CEC Translated Message': Sending state 'TV to Playback 2: Reporting power status (Power On)'
[13:40:35][D][hdmi_cec:089]: frame received: 80:0D
[13:40:35][D][text_sensor:064]: 'HDMI CEC Raw Message': Sending state '80:0D'
[13:40:35][D][text_sensor:064]: 'HDMI CEC Translated Message': Sending state 'Playback 2 to TV: I am the active source'
[13:41:00][D][text_sensor:064]: 'HDMI CEC Raw Message': Sending state '40:9D:44:00'
[13:41:00][D][text_sensor:064]: 'HDMI CEC Translated Message': Sending state 'Playback 1 to TV: Inactive source with physical address (4.4.0.0)'
```

DIYtechie, 2025-06-07:
Since I could not get the OSD names to show perfectly every time, I simplified. 

This seems to be robust (cec decoding including OSD, but OSD is not used for device names):

PS I have suggested this to be used as an example config in my new pull request for the readme.

```
hdmi_cec:
  ...

  on_message:
      
      #CEC message decoder (human-readable translation)
    - then:
        - lambda: |-
            std::vector<uint8_t> frame;
            frame.push_back((source << 4) | (destination & 0xF));
            frame.insert(frame.end(), data.begin(), data.end());
            std::string raw = hdmi_cec::bytes_to_string(frame);
            id(cec_raw_message).publish_state(raw);

            // — Helper functions —
            auto toHex = [&](uint8_t b) {
              char buf[3]; sprintf(buf, "%02X", b);
              return std::string(buf);
            };

            auto getDeviceName = [&](uint8_t nib) {
              static const char* names[] = {
                "TV","Recorder 1","Recorder 2","Tuner 1","Playback 1",
                "Audio system","Tuner 2","Tuner 3","Playback 2","Recorder 3",
                "Tuner 4","Playback 3","Reserved 1","Reserved 2","Free Use","Broadcast"
              };
              return std::string(names[nib & 0xF]);
            };
            
            // - Insert your custom device names here -
            auto applyCustom = [&](uint8_t addr, const std::string &fallback) -> std::string {
              switch (addr) {
                case 0x4: return "Playback 1";      // Playback 1
                case 0x8: return "Playback 2";      // Playback 2
                case 0xB: return "Playback 3";      // Playback 3
                default:  return fallback;
              }
            };

            // — Parse source and destination —
            uint8_t first = frame[0], src = first >> 4, dst = first & 0xF;
            std::string dev_src = applyCustom(src, getDeviceName(src));
            std::string dev_dst = applyCustom(dst, getDeviceName(dst));
            std::string prefix = dev_src + (dst == 0xF
              ? " broadcasting:"
              : " to " + dev_dst + ":");

            uint8_t opcode = frame.size() > 1 ? frame[1] : 0x00;
            uint8_t b2     = frame.size() > 2 ? frame[2] : 0x00;
            uint8_t b3     = frame.size() > 3 ? frame[3] : 0x00;

            std::string phys_addr = std::to_string((b2 & 0xF0) >> 4) + "." +
                                    std::to_string(b2 & 0x0F) + "." +
                                    std::to_string((b3 & 0xF0) >> 4) + "." +
                                    std::to_string(b3 & 0x0F);

            int volume = static_cast<int>(b2) - 2;
            std::string action;

            switch (opcode) {
                case 0x0D:
                case 0x04:
                  action = "I am the active source";
[... 112 lines cut ...]
                    : b2 == 0x02 ? " (Waking up)"
                    : b2 == 0x03 ? " (Power Off)"
                    : " (Unknown: " + toHex(b2) + ")");
                  break;
                default:
                  action = "Unknown action (" + raw + ")";
                  break;
            }
            
            // — Publish the human-readable translation —
            std::string translated = prefix + " " + action;
            id(cec_translated_message).publish_state(translated);

text_sensor:
  - platform: template
    name: "HDMI CEC Raw Message"
    id: cec_raw_message
    update_interval: never

  - platform: template
    name: "HDMI CEC Translated Message"
    id: cec_translated_message
    update_interval: never

```

Minims, 2025-06-07:
Thx, i'll try your last version :-)