Palakis/esphome-native-hdmi-cec #17: [Bug]: ESP WROOM-32 - Boot make TV OFF

Minims, 2025-03-15:
### The problem

Hello,

I'm using this projet for a while and it's really great.

I have an issue that i cannot fix. My TV is going OFF while ESP is Booting / ReBooting
I have my CEC PIN connected to GPIO33.
I have tried to pullup pin 33 with 27k resistor but it does not change anything.

Any idea about how i can fix this issue ?  Thanks.

### Which version of ESPHome was used when this happened?

2023.2.2

### What platform are you using?

ESP32

### Board

ESP-WROOM32

### Example YAML snippet

```yaml
hdmi_cec:
  # Pick a GPIO pin that can do both input AND output
  pin: 
    number: 33
    mode:
      input: true
      output: true
      pullup: false
      pulldown: true

  # The address can be anything you want. Use 0xF if you only want to listen to the bus and not act like a standard device
  address: 0x4

  # Physical address of the device. In this case: 4.0.0.0 (HDMI4 on the TV)
  # DDC support is not yet implemented, so you'll have to set this manually.
  physical_address: 0x4000

  # The name that will we displayed in the list of devices on your TV/receiver
  osd_name: "TV Gateway"

  # By default, promiscuous mode is disabled, so the component only handles directly-address messages (matching
  # the address configured above) and broadcast messages. Enabling promiscuous mode will make the component
  # listen for all messages (both in logs and the on_message triggers)
  promiscuous_mode: true
```

### Anything in the logs that might be useful?

```txt

```

JosVanEijndhoven, 2025-03-15:
It seems that in your YAML code the pin mode is incorrect. It should better be:
```
      input: true
      output: true
      pullup: true
      open_drain: true
```
(So, do NOT specify `pulldown: true`)

Minims, 2025-03-15:
@JosVanEijndhoven Thanks, it seems OK, i also found that `restore_mode` in switchs must be set to `DISABLED`

closed 2025-03-15 by Minims