Palakis/esphome-native-hdmi-cec #21: CEC works for powering OFF, but not for ON

ZyXELis, 2025-06-02:
Hi,

I have setup slwf-08 hdmi device for cec, which uses your repository config in esphome, default settings, fixed gpio14 as per the seller's repo, plugged into hdmi4 (LG oled C4) and physical_address adjusted accordingly:

> hdmi_cec:
>   pin: GPIO14
>   address: 0x05
>   physical_address: 0x4000

Added into home assistant etc. and when I press an off button in the integration - all is ok, it powers off the TV, the log:
> [12:06:20][D][button:010]: 'Turn everything off' Pressed.
> [12:06:20][D][hdmi_cec:217]: sending frame: 5F:36
> [12:06:20][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [12:06:20][D][hdmi_cec:089]: frame received: 0F:87:00:E0:91
> [12:06:20][D][hdmi_cec:089]: frame received: 0F:36
> [12:06:20][D][main:289]: Got Standby command
> [12:06:20][D][hdmi_cec:089]: frame received: 0F:90:01
> [12:06:21][D][hdmi_cec:089]: frame received: 4F:90:01

but when I press the power-on button in hassio - nothing is happening, the TV is still OFF, the log:

> [12:07:09][D][button:010]: 'Turn everything on' Pressed.
> [12:07:09][D][hdmi_cec:217]: sending frame: 5F:04
> [12:07:09][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged

I have a google's streamer connected to HDMI1, which powers-on the TV via CEC normally (I have tried to swap the devices and changing config to different HDMI port with the same result...). Although I can get away with powering it off only - I want to make it work correctly if at all possible. Thanks!

Here's the log from the slwf-8 when I power-on the streamer (streamer - HDMI1, slwf-8 - HDMI4):

> [16:59:49][D][hdmi_cec:089]: frame received: 0F:87:00:E0:91
> [16:59:59][D][hdmi_cec:089]: frame received: 0F:87:00:E0:91
> [17:00:09][D][hdmi_cec:089]: frame received: 0F:87:00:E0:91
> [17:00:17][D][hdmi_cec:089]: frame received: 4F:A6:06:10:40:00
> [17:00:18][D][hdmi_cec:089]: frame received: 4F:84:10:00:04
> [17:00:18][D][hdmi_cec:089]: frame received: 4F:87:00:1A:11
> [17:00:18][D][hdmi_cec:089]: frame received: 45:7D
> [17:00:18][D][hdmi_cec:217]: sending frame: 54:00:7D:00
> [17:00:18][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [17:00:18][W][component:257]: Component hdmi_cec took a long time for an operation (110 ms).
> [17:00:18][W][component:258]: Components should block for at most 30 ms.
> [17:00:19][D][hdmi_cec:089]: frame received: 4F:82:10:00
> [17:00:19][D][hdmi_cec:089]: frame received: 4F:A6:06:10:40:00
> [17:00:20][D][hdmi_cec:089]: frame received: 4F:84:10:00:04
> [17:00:20][D][hdmi_cec:089]: frame received: 4F:87:00:1A:11
> [17:00:20][D][hdmi_cec:089]: frame received: 45:7D
> [17:00:20][D][hdmi_cec:217]: sending frame: 54:00:7D:00
> [17:00:20][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [17:00:20][D][hdmi_cec:089]: frame received: 4F:90:00
> [17:00:23][D][hdmi_cec:084]: ping received: 0x4 -> 0x5
> [17:00:23][D][hdmi_cec:084]: ping received: 0x4 -> 0x5
> [17:00:24][D][hdmi_cec:089]: frame received: 45:83
> [17:00:24][D][hdmi_cec:217]: sending frame: 5F:84:40:00:05
> [17:00:24][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [17:00:24][W][component:257]: Component hdmi_cec took a long time for an operation (141 ms).
> [17:00:24][W][component:258]: Components should block for at most 30 ms.
> [17:00:25][D][hdmi_cec:089]: frame received: 45:46
> [17:00:25][D][hdmi_cec:217]: sending frame: 54:47:48:44:4D:49:2D:43:45:43
> [17:00:25][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [17:00:25][W][component:257]: Component hdmi_cec took a long time for an operation (253 ms).
> [17:00:25][W][component:258]: Components should block for at most 30 ms.
> [17:00:25][D][hdmi_cec:089]: frame received: 0F:84:00:00:00
> [17:00:26][D][hdmi_cec:089]: frame received: 45:8C
> [17:00:26][D][hdmi_cec:217]: sending frame: 54:00:8C:00
> [17:00:26][D][hdmi_cec:242]: HDMICEC::send(): frame not sent: No Ack received
> [17:00:26][D][hdmi_cec:242]: HDMICEC::send(): frame not sent: No Ack received
> [17:00:26][D][hdmi_cec:242]: HDMICEC::send(): frame not sent: No Ack received
> [17:00:26][D][hdmi_cec:242]: HDMICEC::send(): frame not sent: No Ack received
> [17:00:26][D][hdmi_cec:242]: HDMICEC::send(): frame not sent: No Ack received
> [17:00:26][E][hdmi_cec:249]: HDMICEC::send(): send failed after five attempts
> [17:00:26][D][hdmi_cec:089]: frame received: 0F:84:00:00:00
[... 58 lines cut ...]
> [17:00:32][D][hdmi_cec:089]: frame received: 4F:A6:06:10:40:00
> [17:00:32][D][hdmi_cec:089]: frame received: 4F:84:10:00:04
> [17:00:32][D][hdmi_cec:089]: frame received: 4F:87:00:1A:11
> [17:00:33][D][hdmi_cec:089]: frame received: 45:7D
> [17:00:33][D][hdmi_cec:217]: sending frame: 54:00:7D:00
> [17:00:33][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [17:00:33][D][hdmi_cec:089]: frame received: 4F:82:10:00
> [17:00:34][D][hdmi_cec:089]: frame received: 4F:82:10:00
> [17:00:37][D][hdmi_cec:084]: ping received: 0x4 -> 0x5
> [17:00:38][D][hdmi_cec:089]: frame received: 45:46
> [17:00:38][D][hdmi_cec:217]: sending frame: 54:47:48:44:4D:49:2D:43:45:43
> [17:00:38][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [17:00:38][D][hdmi_cec:089]: frame received: 45:8C
> [17:00:38][D][hdmi_cec:217]: sending frame: 54:00:8C:00
> [17:00:38][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [17:00:38][D][hdmi_cec:089]: frame received: 45:8F
> [17:00:38][D][hdmi_cec:217]: sending frame: 54:90:00
> [17:00:38][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [17:00:40][I][safe_mode:041]: Boot seems successful; resetting boot loop counter
> [17:00:40][D][hdmi_cec:089]: frame received: 0F:87:00:E0:91
> [17:00:44][D][hdmi_cec:084]: ping received: 0x0 -> 0x5
> [17:00:44][D][hdmi_cec:089]: frame received: 05:8C
> [17:00:44][D][hdmi_cec:217]: sending frame: 50:00:8C:00
> [17:00:44][D][hdmi_cec:239]: HDMICEC::send(): frame sent and acknowledged
> [17:00:54][D][hdmi_cec:089]: frame received: 0F:87:00:E0:91

DIYtechie, 2025-07-10:
Is this something that should be added to the documentation? Like [here](https://github.com/Palakis/esphome-native-hdmi-cec?tab=readme-ov-file#-2-add-template-buttons-to-send-cec-commands)?

I would think that there are power on commands that work for all TV brands.

I see that Jos uses this for his Sony

```
      on_press:
        - hdmi_cec.send:
            destination: 0x0  # to TV
            data: [0x44, 0x6D]  # UI command, "Power On"

```

and ZyXELis uses this for LG tv

```
      on_press:
        - hdmi_cec.send:
            destination: 0x0  # to TV
            data: [0x0D]

```
 and I use this for my Samsung

```
    on_press:
      hdmi_cec.send:
        source: 1
        destination: 0
        data: [0x04]
```

Might any of these work for all of the TV's?

(Don't need source, so the example should probably be updated)

JosVanEijndhoven, 2025-07-10:
The CEC standard says:
```
<Image View On> 0x04 
Turn on (if not on). If in Text Display state then the TV enters Image Display state.
Note: Should not change TV menu or PIP status.
```
```
<Text View On> 0x0D
As <Image View On>, but should remove PIPs and menus from the screen.
The TV enters Image Display state regardless of its previous state.
```
The standard also says that the above two are mandatory to implement for TVs.
```
UI Functions: 0x6D  Power On Function
Puts the device into the On (non-Standby) state. If repeated, the device stays in the active state
...
In order to deterministically change the power status of the target device,
it is recommended to use the relevant deterministic functions 0x6D, 0x6C or 0x6B
```
I don't see any 'mandatory' in the standard on this `[0x44, 0x6D]`   :-(