Palakis/esphome-native-hdmi-cec #25: [Bug]: Advanced Example Code wont compile

Keith-H-T, 2025-06-24:
### The problem

Compiling .pioenvs/yamaha-hdmi/cmock/CMock/src/cmock.c.o
Compiling .pioenvs/yamaha-hdmi/console/commands.c.o
Archiving .pioenvs/yamaha-hdmi/esp-idf/bootloader_support/libbootloader_support.a
Compiling .pioenvs/yamaha-hdmi/console/esp_console_common.c.o
Archiving .pioenvs/yamaha-hdmi/esp-idf/cmock/libcmock.a
Compiling .pioenvs/yamaha-hdmi/console/split_argv.c.o
Compiling .pioenvs/yamaha-hdmi/console/linenoise/linenoise.c.o
Compiling .pioenvs/yamaha-hdmi/console/esp_console_repl_chip.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_cmd.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_date.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_dbl.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_dstr.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_end.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_file.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_hashtable.c.o
/config/esphome/yamaha-hdmi.yaml: In lambda function:
/config/esphome/yamaha-hdmi.yaml:84:24: error: 'bytes_to_string' is not a member of 'esphome::hdmi_cec'
   84 |             return hdmi_cec::bytes_to_string(full_frame);
      |                        ^~~~~~~~~~~~~~~
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_int.c.o
/config/esphome/yamaha-hdmi.yaml: In lambda function:
/config/esphome/yamaha-hdmi.yaml:92:35: error: 'bytes_to_string' is not a member of 'esphome::hdmi_cec'
   92 |             std::string raw = hdmi_cec::bytes_to_string(frame);
      |                                   ^~~~~~~~~~~~~~~
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_lit.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_rem.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_rex.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_str.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/arg_utils.c.o
Compiling .pioenvs/yamaha-hdmi/console/argtable3/argtable3.c.o
Compiling .pioenvs/yamaha-hdmi/cxx/cxx_exception_stubs.cpp.o
Compiling .pioenvs/yamaha-hdmi/cxx/cxx_guards.cpp.o
Compiling .pioenvs/yamaha-hdmi/cxx/cxx_init.cpp.o
Compiling .pioenvs/yamaha-hdmi/driver/deprecated/adc_legacy.c.o
Archiving .pioenvs/yamaha-hdmi/esp-idf/console/libconsole.a
Compiling .pioenvs/yamaha-hdmi/driver/deprecated/adc_dma_legacy.c.o
*** [.pioenvs/yamaha-hdmi/src/main.cpp.o] Error 1
========================= [FAILED] Took 56.56 seconds =========================

### Which version of ESPHome was used when this happened?

2025.6.1

### What platform are you using?

ESP32-IDF

### Board

ESP32-S3

### Example YAML snippet

```yaml
esphome:
  name: yamaha-hdmi
  friendly_name: Yamaha HDMI

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: esp-idf

# Enable logging
logger:

ota:
  - platform: esphome
[... 366 lines cut ...]
        destination: 8
        data: [0x44, 0x6D]

  - platform: template
    name: "Turn off Playback device 2"
    on_press:
      hdmi_cec.send:
        # "source" can optionally be set, like if you want to spoof another device's address
        destination: 8
        data: [0x36]
        
  - platform: template
    name: "Playback device 2 play/pause"
    on_press:
      hdmi_cec.send:
        # "source" can optionally be set, like if you want to spoof another device's address
        destination: 8
        data: [0x44, 0x46]
```

### Anything in the logs that might be useful?

```txt

```

JosVanEijndhoven, 2025-06-25:
Hi @Keith-H-T ,
sorry, indeed, the README example is not in sync anymore with the recently updated implementation.
The latest implementation has an integrated message decoder, and has a different 'to_string' method.

While we work on renewing the documentation, you  might try a work-around:

- in your yaml, add the new configuration option to disable the built-in decoder:

```
hdmi_cec:
   ...
   decode_messages: false
```

- *AND* replace a few lines in the example lambda c-code:

```
           std::vector<uint8_t> frame;
           frame.push_back((source << 4) | (destination & 0xF));
           frame.insert(frame.end(), data.begin(), data.end());
           return hdmi_cec::bytes_to_string(frame);
```
should become:
```
           hdmi_cec::Frame frame(source, destination, data);
           return frame.to_string();
```

If needed, a similar code replacement further down in the example.

Keith-H-T, 2025-06-25:
Hi, thanks for the quick response. Has not quite fixed the issue, thrown up a couple of different errors. Happy to wait until the documentation is updated. Regards Keith

Palakis, 2025-07-04:
@JosVanEijndhoven 

> [...] should I keep the (partial) decoder from this example, or remove that and re-use the decoder that now is in the component itself?

I'd go for removing the partial decoder and using the built-in one.

Additionally, now that the `to_string` is a member of `Frame` and can't be called independently, we probably should pass the `Frame` instance in the `on_message` callback.

anishsane, 2025-07-09:
> we probably should pass the Frame instance in the on_message callback.

But we should also have the ability to filter based on the opcodes/src/dest etc.
Alternately, we can pass src, dest, data_string to the on_message callback.

closed 2025-09-15 by github-actions[bot]