Palakis/esphome-native-hdmi-cec #28: [Bug]: Access to invalid address on esphome 8266

mhetzi, 2025-07-08:
### The problem

Hi,
Today i tried updateing the config of my CEC ESP. With current esphome and esphome-native-hdmi-cec this and it instantly crashes to safe mode.

I tried a random older commit ([11c4b9aee7add7879fea3d470851da76142f9e86](https://github.com/Palakis/esphome-native-hdmi-cec/tree/11c4b9aee7add7879fea3d470851da76142f9e86)), with this it works.

### Which version of ESPHome was used when this happened?

2025.6.3

### What platform are you using?

ESP8266

### Board

SMLIGHT.SLWF-08-HDMI

### Example YAML snippet

```yaml

```

### Anything in the logs that might be useful?

```txt
[20:02:14]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[20:02:14]
[20:02:14]Exception (28):
WARNING Exception type: Access to invalid address: LOAD (wild pointer?)
[20:02:14]epc1=0x401001d4 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
WARNING Decoded 0x401001d4: esphome::ISRInternalGPIOPin::digital_read()
[20:02:14]
[20:02:14]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[20:02:14]
[20:02:14]ctx: cont
[20:02:14]sp: 3ffffcc0 end: 3fffffd0 offset: 0150
[20:02:14]3ffffe10:  3fff2cac 3fff01c8 3fff34ac 401003ef  
WARNING Decoded 0x401003ef: esphome::hdmi_cec::HDMICEC::gpio_intr_(esphome::hdmi_cec::HDMICEC*)
[20:02:14]3ffffe20:  3ffffea0 3ffffe90 00000010 3ffe9baf  
[20:02:14]3ffffe30:  40100b2c 00000001 0000000e 40100bb8  
WARNING Decoded 0x40100b2c: interrupt_handler
WARNING Decoded 0x40100bb8: interrupt_handler
[20:02:14]3ffffe40:  00000020 00000106 1c6a7ef9 000185c8  
[20:02:14]3ffffe50:  00000150 3fff0c30 00000001 00000022  
[20:02:14]3ffffe60:  3fffc200 40100af4 3fffc258 4000050c  
WARNING Decoded 0x40100af4: interrupt_handler
[20:02:14]3ffffe70:  40215972 00000030 00000012 ffffffff  
WARNING Decoded 0x40215972: esphome::hdmi_cec::HDMICEC::setup()
[20:02:14]3ffffe80:  4021596e 3fff35f4 4022e004 60000360  
WARNING Decoded 0x4021596e: esphome::hdmi_cec::HDMICEC::setup()
WARNING Decoded 0x4022e004: precache
[20:02:14]3ffffe90:  00000180 00000780 fffffffd 00000000  
[20:02:14]3ffffea0:  4022aa4c 40227f1c 3ffffcc3 3fff34ac  
WARNING Decoded 0x4022aa4c: pinMode
WARNING Decoded 0x40227f1c: HardwareSerial::write(unsigned char const*, unsigned int)
[20:02:14]3ffffeb0:  00000005 3fff34ac 3fff3304 00000030  
[20:02:14]3ffffec0:  3fff42fc 00000038 00000038 40210046  
WARNING Decoded 0x40210046: esphome::api::DeviceInfoResponse::decode_length(unsigned int, esphome::api::ProtoLengthDelimited)
[20:02:14]3ffffed0:  0000000e 0000000a 3fff42f4 3fff42f6  
[20:02:14]3ffffee0:  00002e20 000005c4 3ffe867c 40101036  
WARNING Decoded 0x40101036: umm_free_core at umm_malloc.cpp
[20:02:14]3ffffef0:  00000000 00000000 00051b65 3fff3304  
[20:02:14]3fffff00:  00000000 00000020 3fff42ec 40101384  
WARNING Decoded 0x40101384: free
[20:02:14]3fffff10:  00000000 00000005 3fff3ca4 40222824  
WARNING Decoded 0x40222824: esphome::hdmi_cec::SendAction<>::play()
[20:02:14]3fffff20:  3fff42ec 3fff42ed 3fff42ed 3fff30f8  
[20:02:14]3fffff30:  3fff3e80 3fff3c8c 3fff3ca4 4021f314  
WARNING Decoded 0x4021f314: esphome::Action<>::play_complex()
[20:02:14]3fffff40:  3ffe8c53 3fff0708 3fff3c7c 3fff0704  
[20:02:14]3fffff50:  3ffea21b 00000000 3fff3eec 4021f314  
WARNING Decoded 0x4021f314: esphome::Action<>::play_complex()
[20:02:14]3fffff60:  3fff0001 4021d9de 3fff34ac 4021596e  
WARNING Decoded 0x4021d9de: void std::__stable_sort_adaptive<__gnu_cxx::__normal_iterator<esphome::Component**, std::vector<esphome::Component*, std::allocator<esphome::Component*> > >, esphome::Component**, int, __gnu_cxx::__ops::_Iter_comp_iter<esphome::Application::setup()::{lambda(esphome::Component const*, esphome::Component const*)#1}> >(__gnu_cxx::__normal_iterator<esphome::Component**, std::vector<esphome::Component*, std::allocator<esphome::Component*> > >, __gnu_cxx::__ops::_Iter_comp_iter<esphome::Application::setup()::{lambda(esphome::Component const*, esphome::Component const*)#1}>, esphome::Component**, int, __gnu_cxx::__ops::_Iter_comp_iter<esphome::Application::setup()::{lambda(esphome::Component const*, esphome::Component const*)#1}>) at application.cpp
WARNING Decoded 0x4021596e: esphome::hdmi_cec::HDMICEC::setup()
[20:02:14]3fffff70:  3fff3e80 00000004 3fff0704 4021da79  
WARNING Decoded 0x4021da79: esphome::Application::setup()
[20:02:14]3fffff80:  0000000e 0000000e 3fff4294 00000001  
[20:02:14]3fffff90:  00000018 3fff0810 3fff0704 3fff0860  
[20:02:14]3fffffa0:  feefeffe feefeffe feefeffe 3fff0c58  
[20:02:14]3fffffb0:  3fffdad0 00000000 3fff0c2c 4022a320  
WARNING Decoded 0x4022a320: loop_wrapper() at core_esp8266_main.cpp
[20:02:14]3fffffc0:  feefeffe feefeffe 3fffdab0 401007c1  
WARNING Decoded 0x401007c1: cont_wrapper
[20:02:14]<<<stack<<<
```

Palakis, 2025-07-19:
Hi @mhetzi, could you please provide the YAML file you used to build that device? (don't forget to omit passwords, wifi SSIDs and other secrets)

mhetzi, 2025-07-19:
ok:
```

substitutions:
  name: wh3-hdmi-cec
  friendly_name: "WH3 CEC"

esphome:
  name: "${name}"
  friendly_name: "${friendly_name}"
#  platformio_options:
#    board_buid.f_cpu: 160000000L
  name_add_mac_suffix: true
  project:
    name: SMLIGHT.SLWF-08-HDMI
    version: "1.0-Palakis"
  on_boot:
    - priority: 0
      then:
        - delay: 5s
        - button.press: btn_give_pwr_state

esp8266:
  board: esp12e
  framework: 
    version: recommended

wifi:
  domain: ".iot"
  power_save_mode: HIGH
  fast_connect: false
  use_address: wh3-hdmi-control-palakis-e18094.iot
  networks:
    ssid: !secret wifi_ssid
    password: !secret wifi_psk
    hidden: false

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "esphome BackupAP"
    password: !secret ap_psk

# Enable Home Assistant API
api:
  encryption:
    key: <KEY>
  services:
    - service: hdmi_cec_send
      variables:
        cec_destination: int
        cec_data: int[]
      then:
        - hdmi_cec.send:
            destination: !lambda "return static_cast<unsigned char>(cec_destination);"
            data: !lambda "std::vector<unsigned char> charVector; for (int i : cec_data) { charVector.push_back(static_cast<unsigned char>(i)); } return charVector;"

ota:
  - platform: esphome
    password: !secret api_pw
    
safe_mode:
    reboot_timeout: 30min

time:
  - platform: homeassistant
    id: htime
    timezone: Europe/Vienna
    on_time:
      - seconds: 0
        minutes: /1
        then:
          - button.press: btn_give_pwr_state

# Enable logging
logger:
  level: DEBUG
  logs:
    text_sensor: INFO

external_components:
#  - source: github://johnboiles/esphome-hdmi-cec
  - source: 
      type: git
      url: https://github.com/Palakis/esphome-hdmi-cec
      ref: 11c4b9aee7add7879fea3d470851da76142f9e86
    refresh: 1h

dashboard_import:
  package_import_url: github://smlight-tech/slwf-08/esphome-configs/slwf-08-Palakis.yaml

hdmi_cec:
  pin: GPIO14
  address: 0x05
  physical_address: 0x2000
  osd_name: "esphome" # Optional. Defaults to "esphome"  
  promiscuous_mode: false # Optional. Defaults to false  
  monitor_mode: false # Optional. Defaults to false  
  on_message:
    - opcode: 0x36 # opcode for "Standby"
      then:
        - logger.log: "Got Standby command"
        - switch.template.publish:
            id: sw_tv_on_off
            state: OFF
        # Respond to "Menu Request" (not required, example purposes only)
    - opcode: 0x8D
      then:
        hdmi_cec.send:
          # both "destination" and "data" are templatable
          destination: !lambda return source;
          data: [0x8E, 0x01] # 0x01 => "Menu Deactivated" 
    - opcode: 0x90 # opcode for "Standby"
      then:
        logger.log: "Report Power Status"
    - data: [0x90, 0x00]
      then:
        - logger.log: "TV Power on"
        - switch.template.publish:
            id: sw_tv_on_off
            state: ON
    - data: [0x90, 0x01]
      then:
        - logger.log: "TV Power off"
        - switch.template.publish:
            id: sw_tv_on_off
            state: OFF

button:
  - platform: template
    name: "Turn everything on"
    on_press:
      - hdmi_cec.send:
          destination: 0xF # or 0xF for Broadcast
          data: [0x04] # "Image View on
  - platform: template
    name: "Turn everything off"
    id: btn_all_off
    on_press:
      - hdmi_cec.send:
          destination: 0xF
          data: [0x36] # "Standby" opcode
  - platform: template
    name: "Turn active source"
    on_press:
      - hdmi_cec.send:
          destination: 0xF # Broadcast
          data: [0x82] # "Power on" opcode
  - platform: template
    name: "Mute"
    on_press:
      - hdmi_cec.send:
          destination: 0xF # Broadcast
          data: [0x43] # "Mute" opcode          
  - platform: template
    name: "Volume Down"
    on_press:
      - hdmi_cec.send:
          destination: 0xF # Broadcast
          data: [0x42] # "Volume Down" opcode
  - platform: template
    name: "Volume up"
    on_press:
      - hdmi_cec.send:
          destination: 0xF # Broadcast
          data: [0x41] # "Volume up" opcode   
  - platform: template
    name: "Image View on"
    id: btn_image_view_on
    on_press:
      - hdmi_cec.send:
          destination: 0x0 # TV
          data: [0x04] # "Image View on
  - platform: template
    name: "Give Power Status"
    id: btn_give_pwr_state
    on_press:
      - hdmi_cec.send:
          destination: 0x0 # TV
          data: [0x8F] # Query Power Status
  

switch:
  - platform: template
    name: "TV Eingeschaltet"
    id: sw_tv_on_off
    turn_on_action:
      then:
        - button.press: btn_image_view_on
        - button.press: btn_give_pwr_state
    turn_off_action:
      then:
        - button.press: btn_all_off
        - button.press: btn_give_pwr_state

binary_sensor:
  - platform: status
    name: "WH3 CEC Status"
    id: myState
```

closed 2025-09-25 by github-actions[bot]