Palakis/esphome-native-hdmi-cec #29: [Feature]: Binary Sensor or Switch Component

virtualJonesie, 2025-07-11:
**Describe what new feature you would like**

A switch (or binary sensor which could be used to build a template switch in Home Assistant) to allow you to know the state of the HDMI device and control it.

**Please describe your use case for this feature and alternatives you've tried:**

I would like to use the status from a switch, or binary sensor, to control other aspects of Home Assistant, such as automations.

**Additional context**

No

virtualJonesie, 2025-07-11:
I have reviewed the examples on this repo, as well as the documentation for the ESPHome Switch and Binary Sensor components, and I have had no luck getting this to work.

DIYtechie, 2025-11-12:
This would still be cool to have at some point in the future 🤓 But no rush for my sake.