Palakis/esphome-native-hdmi-cec #31: [Bug]: CEC module does not seem to be sending responses to incoming queries automatically.

russ-sentient, 2025-08-16:
### The problem

When I watch the log, I am seeing the TV spam Give OSD Name.  I have monitor mode off, the physical address is set to the port the HDMI is plugged into, and the device is set as specific use 0xE.  I think that some devices start to shadow blacklist the esp device if it does not respond to queries, but I cannot confirm.

I looked through the github src and see that it should be trying to respond to these in the backend, is there a reason it may not be happening?  Should I write some logic to respond to these messages in my esphome config?

`[13:51:26][D][hdmi_cec:102]: [received] 0E:46 => TV to SpecificUse: <Give OSD Name>[]
[13:51:30][D][hdmi_cec:102]: [received] 0E:46 => TV to SpecificUse: <Give OSD Name>[]
[13:51:34][D][hdmi_cec:102]: [received] 0E:46 => TV to SpecificUse: <Give OSD Name>[]
[13:51:38][D][hdmi_cec:102]: [received] 0E:46 => TV to SpecificUse: <Give OSD Name>[]
[13:51:42][D][hdmi_cec:102]: [received] 0E:46 => TV to SpecificUse: <Give OSD Name>[]
[13:51:46][D][hdmi_cec:102]: [received] 0E:46 => TV to SpecificUse: <Give OSD Name>[]`

### Which version of ESPHome was used when this happened?

2025.7.1

### What platform are you using?

ESP32-IDF

### Board

esp32-s3-devkitc-1

### Example YAML snippet

```yaml

```

### Anything in the logs that might be useful?

```txt

```

russ-sentient, 2025-08-16:
I ended up throwing this in for now to respond to the two queries that were being spammed on the bus to keep the chatter down.

`
  on_message:

    - opcode: 0x46
      then:
        hdmi_cec.send:
          # both "destination" and "data" are templatable
          destination: !lambda return source;
          data: [0x47, 0x43, 0x45, 0x43, 0x20, 0x5a, 0x31] #CEC Z1

    - opcode: 0x8C
      then:
        hdmi_cec.send:
          # both "destination" and "data" are templatable
          destination: !lambda return source;
          data: [0x87,0x0f,0x42,0x3f] #VENDOR ID 999999
`

Minims, 2025-09-02:
I have the same issue, will try your workaroud