Palakis/esphome-native-hdmi-cec #32: [Feature]: Configurable Physical Address

z06Drew, 2025-08-29:
**Describe what new feature you would like**

Is it possible to read the physical address from Home Assistant, like a configuration value?

**Please describe your use case for this feature and alternatives you've tried:**

I dont want to have to upload new code just to move the dongle to a different HDMI port

**Additional context**

JosVanEijndhoven, 2025-08-30:
Such 'feature' is already possible in your application yaml, and so is not really part of the 'hdmi-cec' library component.

For example, from my own configuration:
```
number:
  - platform: template
    id: hdmi_port
    name: "Hdmi Port Nr"
    icon: "mdi:video-input-hdmi"
    min_value: 1
    max_value: 6
    step: 1
    optimistic: true
    initial_value: 3
    restore_value: true
    on_value:
      then:
        - logger.log:
            format: "Update HDMI port on TV to %.0f"
            args: [ id(hdmi_port).state ]
            level: "INFO"
        - lambda: !lambda |-
            const uint16_t port = std::lround(id(hdmi_port).state);
            id(cec).set_physical_address(((port & 0x7) << 12));  // we connect directly to the TV, no device in between
```
Note that after modifying the port, you probably need to power-cycle the device for a restart, so it announces itself again on the bus with its new address.

In this example the `id(cec)` refers to the `hdmi-cec` component declaration such as in:
```
hdmi_cec:
  id: cec
  pin: GPIO....
```

JosVanEijndhoven, 2026-01-09:
Yes, of course, feel free to use the above code fragment as starting point for ddc access, probably as part of the cec module.
I have successfully tested it on three different TV brands. The many ESP_LOGs helped me to observe the actual behavior,
but are probably overkill in a final implementation.

Note the need for an i2c bus level shifter to 5V. These level shifters simultaneously protect the cec module against voltages applied to the microcontroller io pins when its own power would be off.