Palakis/esphome-native-hdmi-cec #33: [Bug]: ESP32C3 crashing

TheAndi, 2025-08-30:
### The problem

When sending or receiving a few commands, ESP is crashing regularly.
I now managed to capture a serial log when this happened 

```
[02:15:24][D][button:010]: 'D-Pad Left (TV)' Pressed.
[02:15:24][D][hdmi_cec:230]: [sending] 40:44:03 => PlaybackDev1 to TV: <User Control Pressed>[Left]
[02:15:29]E (118932) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:
[02:15:29]E (118932) task_wdt:  - loopTask (CPU 0)
[02:15:29]E (118932) task_wdt: Tasks currently running:
[02:15:29]E (118932) task_wdt: CPU 0: IDLE
[02:15:29]E (118932) task_wdt: Aborting.
[02:15:29]
[02:15:29]
[02:15:29]Core  0 register dump:
[02:15:29]MEPC    : 0x40384b0a  RA      : 0x403889b6  SP      : 0x3fc9e9a0  GP      : 0x3fc93600  
[02:15:29]TP      : 0x3fc9e9e0  T0      : 0x00000000  T1      : 0x40384c6e  T2      : 0x00000000  
[02:15:29]S0/FP   : 0x3fc9a000  S1      : 0x3fc965f4  A0      : 0x00000001  A1      : 0x3fc9e294  
[02:15:29]A2      : 0x00001154  A3      : 0x00000000  A4      : 0x00000000  A5      : 0x00000000  
[02:15:29]A6      : 0x00000000  A7      : 0x00000000  S2      : 0x3fc9a000  S3      : 0x3fc9e294  
[02:15:29]S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
[02:15:29]S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
[02:15:29]T3      : 0x00000000  T4      : 0x00000000  T5      : 0x00000000  T6      : 0x00000000  
[02:15:29]MSTATUS : 0x00000000  MTVEC   : 0x00000000  MCAUSE  : 0x00000000  MTVAL   : 0x00000000  
[02:15:29]MHARTID : 0x00000000  
[02:15:29]
[02:15:29]Stack memory:
[02:15:29]3fc9e9a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[02:15:29]3fc9e9c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
[02:15:29]3fc9e9e0: 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x3fc9e920 0x00000000 0x3fc9664c 0x3fc9664c 0x3fc9e9ec
[02:15:29]3fc9ea00: 0x3fc96644 0x00000019 0x00000000 0x00000000 0x3fc9e9ec 0x00000000 0x00000000 0x3fc9e3e8
[02:15:29]3fc9ea20: 0x454c4449 0x00000000 0x00000000 0x00000000 0x3fc9e9e0 0x00000000 0x00000000 0x00000000
[02:15:29]3fc9ea40: 0x00000000 0x00000000 0x3fc9ab68 0x3fc9abd0 0x3fc9ac38 0x00000000 0x00000000 0x00000001
[02:15:29]3fc9ea60: 0x00000000 0x00000000 0x00000000 0x42049c06 0x00000000 0x00000000 0x00000000 0x00000000
[02:15:29]3fc9ea80: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[02:15:29]3fc9eaa0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[02:15:29]3fc9eac0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[02:15:29]3fc9eae0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[02:15:29]3fc9eb00: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[02:15:29]3fc9eb20: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000200 0x0000000c
[02:15:29]3fc9eb40: 0x094a000a 0x00000000 0x3fc9ebbc 0x00000054 0x00000000 0x3fc9eb50 0x00000000 0x00000000
[02:15:29]3fc9eb60: 0x00000000 0x3fc9eb68 0xffffffff 0x3fc9eb68 0x3fc9eb68 0x00000000 0x3fc9eb7c 0xffffffff
[02:15:29]3fc9eb80: 0x3fc9eb7c 0x3fc9eb7c 0x00000001 0x00000001 0x00000000 0x0c00ffff 0x00000000 0x15863635
[02:15:29]3fc9eba0: 0x3fc9eb48 0x00000010 0x3fc965a4 0x3fca3588 0x00000000 0x00000001 0x0000000c 0x084e0008
[02:15:29]3fc9ebc0: 0x00000000 0x3fca3578 0x0000000c 0x3c0d62a8 0x00000001 0x3fc9ebc4 0x00000054 0x00000000
[02:15:29]3fc9ebe0: 0x3fc9ebdc 0x00000000 0x00000000 0x00000000 0x3fc9ebf4 0xffffffff 0x3fc9ebf4 0x3fc9ebf4
[02:15:29]3fc9ec00: 0x00000000 0x3fc9ec08 0xffffffff 0x3fc9ec08 0x3fc9ec08 0x00000001 0x00000001 0x00000000
[02:15:29]3fc9ec20: 0xb800ffff 0x00000000 0x625c11d0 0x3fc9ebd4 0x00000028 0x3c0da358 0x3fca139c 0x3fcaa194
[02:15:29]3fc9ec40: 0x00000003 0x3fc9ee80 0x3fc9ee80 0x00000001 0x3fc9ecb8 0x3fc9ecb8 0x00000001 0x00000054
[02:15:29]3fc9ec60: 0x00000000 0x3fc9ec60 0x00000000 0x00000000 0x00000000 0x3fc9ec78 0xffffffff 0x3fc9ec78
[02:15:29]3fc9ec80: 0x3fc9ec78 0x00000000 0x3fc9ec8c 0xffffffff 0x3fc9ec8c 0x3fc9ec8c 0x00000001 0x00000001
[02:15:29]3fc9eca0: 0x00000000 0x0b00ffff 0x00000000 0x4f2e1a60 0x3fc9ec58 0x0000001c 0x3c0da324 0x00000000
[02:15:29]3fc9ecc0: 0x00000000 0x3fc9ee48 0x3fc9ee48 0x3fc9a454 0x3fc9ecb0 0x0000002c 0x000f0000 0x000f0000
[02:15:29]3fc9ece0: 0x00000000 0x00000000 0x00000000 0x00000012 0x00000000 0x00000000 0x00000000 0x3fc9ed08
[02:15:29]3fc9ed00: 0x3fc99d4c 0x0000002c 0x00800000 0x00800000 0x00000000 0x00000000 0x00000000 0x00000012
[02:15:29]3fc9ed20: 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9ecfc 0x0000001c 0x00000000 0x3fca133c
[02:15:29]3fc9ed40: 0x68707365 0x00656d6f 0x00000000 0x00000000 0x3fc9ed01 0x0000000c 0x3fc9d8c8 0x3fc9d92c
[02:15:29]3fc9ed60: 0x3fc9d984 0x00000034 0x3fc945f8 0x00000001 0x00000000 0x00009000 0x00002000 0x00001000
[02:15:29]3fc9ed80: 0x6461746f 0x00617461 0x00000000 0x00000000 0x00000000 0x00000000 0x3fc9eda0 0x00000034
[02:15:29]
[02:15:29]
[02:15:29]
[02:15:29]ELF file SHA256: 4a7a86c1f
[02:15:29]
[02:15:29]Rebooting...
[02:15:29]ESP-ROM:esp32c3-api1-20210207
[02:15:29]Build:Feb  7 2021
[02:15:29]rst:0xc (RTC_SW_CPU_RST),boot:0xf (SPI_FAST_FLASH_BOOT)
[02:15:29]Saved PC:0x403814f6
[... 191 lines cut ...]
  - platform: template
    name: "D-Pad Left (TV)"
    on_press:
      - hdmi_cec.send:
          destination: 0x0
          data: [0x44, 0x03]
```

### Anything in the logs that might be useful?

```txt
also I regulary get warnings like these:

(time not related)
02:16:56 | [W] | [component:407] | hdmi_cec took a long time for an operation (405 ms)
02:16:56 | [W] | [component:408] | Components should block for at most 30 ms

maybe at some point, it's too long and triggers the watchdog?
```

### Additional information

Is there any chance this can be fixed? 

Thanks very much in advance. I happily test new solutions!

JosVanEijndhoven, 2025-08-30:
The uart mode is in use by myself for nearly a year now, and is separately tested on 4 different microcontrollers, each with both the arduino and esp-idf framework build types. The code seems stable from my point of view.

nareso, 2025-09-29:
Dear JosVanEijndhoven

We look forward to the release of uart mode.
The hdmi_cec component seems to be stuck in review, so please update the check.

JosVanEijndhoven, 2025-10-01:
Yes. This summer I was too busy with another project. I am just starting again to pull this off. I fact, the review did not yet start, as I was still preparing the documentation and tests...

closed 2026-02-20 by github-actions[bot]

ntzb, 2026-09-06:
Late to this, but it might be the same thing I've been chasing.

`send_high_and_test_` computes its waits as `CONSTANT - (micros() - start)`, unsigned. If the bit has already overrun the deadline that underflows, and `delay_microseconds_safe` would then sit in `vTaskDelay` for ~71 minutes - long enough for the watchdog to take the loop task out mid-send. Your `task_wdt loopTask` dump right after a send could be that.

On a C3 here it counted 7 hits over ~65 presses, worst 712us past the deadline. #56 if you want to try it.

nareso, 2026-09-06:
https://github.com/mguaylam/esphome-cec is one of solution without crash.
I used it more than one month and have very stable.