Palakis/esphome-native-hdmi-cec #34: [Bug]: Play/Pause toggle

exetico, 2025-08-31:
### The problem

Hi,

I'm experiencing that "2 play/pause" does only pause my Apple TV.

The ESP connects to my LG TV, and I use a Apple TV for most media-playback sessions.

Using the `2 play/pause` can only pause the playback.

```yaml
  - platform: template
    name: "Playback device 2 play/pause"
    on_press:
      hdmi_cec.send:
        # "source" can optionally be set, like if you want to spoof another device's address
        destination: 8
        data: [0x44, 0x46]
```

I've not looked that much into it, but the device 1 play/pause does not work on the Apple TV, at all.

### Which version of ESPHome was used when this happened?

2025.8.2

### What platform are you using?

ESP8266

### Board

NodeMCU

### Example YAML snippet

```yaml
I've used the advance example, but the topic relates to:

  - platform: template
    name: "Playback device 2 play/pause"
    on_press:
      hdmi_cec.send:
        # "source" can optionally be set, like if you want to spoof another device's address
        destination: 8
        data: [0x44, 0x46]
```

### Anything in the logs that might be useful?

```txt
ChatGPT mentions that the Apple TV expects a `0x45` at the end, indicating that the user ended the key-press. I've tested it, and it does work:

  - platform: template
    name: "Playback device 2 play/pause"
    on_press:
      - hdmi_cec.send:
          destination: 8
          data: [0x44, 0x44]   # Play/Pause pressed
      - delay: 100ms
      - hdmi_cec.send:
          destination: 8
          data: [0x45]         # Key released

I'm not sure if this will impact the experience for other people, however, it does fix my issue on Apple TV.
```

### Additional information

You'e more than welcome to rename the title, if this isn't a bug, more like a "incompatibility" with the Apple TV, without tweaking the advance settings.

I'ved tested the change today, and now play/pause works as expected on the Apple TV.

DIYtechie, 2025-09-01:
Sorry for not providing more context about the buttons earlier.

I’ve tested the configuration and can confirm:

This works as play/pause for PS4:

```
  - platform: template
    name: "Playback device 2 play/pause"
    on_press:
      hdmi_cec.send:
        # "source" can optionally be set, like if you want to spoof another device's address
        destination: 8
        data: [0x44, 0x46]
```

This works for Apple TV:

```
  - platform: template
    name: "Playback device 1 play/pause"
    on_press:
      hdmi_cec.send:
        # "source" can optionally be set, like if you want to spoof another device's address
        destination: 4
        data: [0x44, 0x44]
```

No delay or repeated presses are needed.

So, both buttons function correctly — they’re simply mapped to different devices.

DIYtechie, 2025-09-01:
In the README example, it would probably have been clearer if I had written “Apple TV” instead of “Playback device 1”, and “PlayStation” instead of “Playback device 2”. Perhaps worth amending?

DIYtechie, 2025-11-12:
Based on the latest PR I think this issue is now resolved. Playback toggle works on Apple TV.